STMicroelectronics/STM32CubeG0 #2: DMAMUX_IdRegisters_TypeDef is undefined

salkinium, 2019-06-04:
**Describe the set-up**
 * arm-none-eabi-gcc 
 * `#include <stm32g0xx.h>`

**Describe the bug**

In every CMSIS device header there is a line like this:
```
#define DMAMUX1_IdRegisters        ((DMAMUX_IdRegisters_TypeDef *) DMAMUX1_IdRegisters_BASE)
```

However, `DMAMUX_IdRegisters_TypeDef` is never declared.

**How To Reproduce**

Just try to use the `DMAMUX_IdRegisters_TypeDef` anywhere in your code, for example like this:
```
#define STM32G071xx 1
#include <stm32g0xx.h>
DMAMUX_IdRegisters_TypeDef type;
```

**Additional context**

I found this issue during debugging when I wanted to read the values of `DMAMUX1_IdRegisters` in GDB, which failed since `DMAMUX_IdRegisters_TypeDef` was unknown. I can read the values manually, but it's quite annoying.

salkinium, 2019-06-06:
cc @CCASTM

CCASTM, 2019-06-07:
Hello Niklas,
Thank you for your finding. 
This definition is obsolete, you can simply remove it.
We will correct this situation in our next publication of the STM32G0 FW Package
With ALi @ALABSTM  
Internal ST Reference 67835
Regards
Christophe

closed 2019-06-07 by CCASTM

ALABSTM, 2019-11-29:
ST Internal Reference: 67835