STMicroelectronics/STM32CubeG0 #4: LL RTC - wrong reference to RTC

saeugetier, 2019-07-21:
`LL_RTC_DisableInitMode(RTCx);` should be used instead of `LL_RTC_DisableInitMode(RTC);`.

There is only one instance of RTC. But in common the RTCx pointer is used in every function. No direct access of RTC is used.

saeugetier, 2019-07-21:
https://github.com/STMicroelectronics/STM32CubeG0/blob/2447e18b9505b881c926e8d6c6fc60d065d85ebc/Drivers/STM32G0xx_HAL_Driver/Src/stm32g0xx_ll_rtc.c#L325

ALABSTM, 2019-07-22:
Dear Timm,

You are absolutely right. This shall be corrected and made available in a future release. Thank you for pointing out this issue.

With regards,

ALABSTM, 2019-07-22:
ST Internal Reference: 69910

CCASTM, 2019-08-21:
Hello Timm,
The correction that you submitted is valid.
We have recorded and corrected our internal software database. The next STM32G0 will include the correction.
If you have no other question, can we close this Issue ?
Regards
Christophe

saeugetier, 2019-08-21:
Hello Christophe,

thanks. That's all. You can close the issue.

Regards
Timm

closed 2019-08-21 by saeugetier