STMicroelectronics/STM32CubeG0 #5: Naming inconsistency for Flash Busy flag

psarvesh, 2020-01-13:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
Discovered an inconsistency in naming when making a pull request in some other project (https://github.com/zephyrproject-rtos/zephyr/pull/20754).

The FLASH_SR_BSY1 flag for flash in STM32CubeG0 package should be called FLASH_SR_BSY to maintain consistency with other micros in STM32 series

ASELSTM, 2020-02-03:
Hi Sarvesh,

The issue you pointed out has been communicated to our development teams. It will be investigated soon. We will be back to you as soon as we get an answer. Thank you for your patience.

With regards,

ALABSTM, 2020-03-12:
Hi Sarvesh,

Thank you for having pointed out this aspect. Actually, the flag has been named FLASH_SR_BSY1 to be compliant with the **reference manual**.

However, it would enhance compatibility with other products if this flag is also aliased FLASH_SR_BSY. This will be done in a future release. We cannot share a date for the moment but we hope this is done soon enough.

With regards,

ALABSTM, 2020-03-12:
ST Internal Reference: 82764

ALABSTM, 2020-04-02:
Hi Sarvesh,

I hope you are doing well. Your request has been submitted to our technical committee who said it is more appropriate to keep the flag name unchanged (i.e. FLASH_SR_BSY**1**).

Indeed, beyond the fact this name is compliant with the reference manual, it shall be left as it is for the need of a future use.

Unfortunately, your request could not be addressed this time. Thank you for your comprehension and thank you again for having reported this point.

Please, allow me to close this issue now. Take care and stay safe.

With regards,

closed 2020-04-02 by ALABSTM