STMicroelectronics/STM32CubeG0 #9: HDMI CEC extra byte sent.

bbooher, 2020-05-19:
**Describe the set-up**
Custom board with G071G8 and required HDMI circuitry.
VS Code with GCC: not a compiler-dependent issue.

**Describe the bug**
An extra byte is sent with CEC commands due to an off-by-one bug in the IRQ Handler in stm32g0xx_hal_cec.c LINE 825. The logical comparison should be with 1 instead of 0 since the decrement has not yet occurred.

**How To Reproduce**
1. A valid HDMI device is needed. Send a CEC command using the ...Transmit_IT function in the HAL.
2. Sending any CEC command longer than 1 byte.
3. A logic analyzer with a CEC decoder is useful to observe the problem.

ASELSTM, 2020-05-20:
Hi @bbooher,

Thank you for reporting this issue. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get an answer. Thank you in advance for your patience.

With regards,

ASELSTM, 2020-05-27:
Hi @bbooher,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-05-27:
ST Internal Reference: 87137

ALABSTM, 2020-10-12:
Same issue reported on STM32CubeF0 [here](https://github.com/STMicroelectronics/STM32CubeF0/issues/5).

ASELSTM, 2021-08-16:
Hi @bbooher,

The issue you reported has been fixed in the frame of version v1.5.0 of the STM32CubeG0 published on GitHub.

Please allow me then to close this thread and thank you again for having reported.

With regards,

closed 2021-08-16 by ASELSTM