STMicroelectronics/STM32CubeG0 #10: Bug exists in FreeRTOS CMSIS RTOS V2 API

bridadan, 2020-07-07:
**Describe the set-up**
 * Custom board based on the STM32G070KBT6
 * STM32CubeIDE 1.3.1 (using STM32CubeG0 1.3.0)

**Describe the bug**
Version 1.3.0 of the STM32CubeG0 includes version 10.0.1 of CMSIS-FreeRTOS. There is a bug in this version that is described  and fixed here: https://github.com/ARM-software/CMSIS-FreeRTOS/issues/12

**How To Reproduce**
Anytime `osEventFlagsSet` is called from an ISR, it fails.

**Additional context**
The first version of CMSIS-FreeRTOS that includes this fix is 10.2.0. Could the FreeRTOS version be updated in a future release?

Thanks very much for providing this library! It is one of the many reasons I keep use ST products 😄

RKOUSTM, 2020-07-13:
Hi @bridadan,

Thank you for your contribution. Your message will be forwarded to our development teams. We will be back to you as soon as we get an answer. 

With regards,

RKOUSTM, 2020-07-22:
Hi @bridadan,

Thank you for your interest. We are planning to update the STM32CubeG0 with the latest Version 1.3.0 of CMSIS-FreeRTOS for future release. We aim to have this package available on Github soon. We hope this addresses your request. In such case this issue could be closed. Thank you for your patience.

With regards,

bridadan, 2020-07-22:
Hi @RKOUSTM, thanks for the update! Good to hear, though just to confirm, the latest version of CMSIS-FreeRTOS would be [_10.3.1_](https://github.com/ARM-software/CMSIS-FreeRTOS/releases/tag/10.3.1), not _1.3.0_. Do you intend to use 10.3.1?

RKOUSTM, 2020-07-23:
Hi @bridadan, 

Indeed, the future release of STM32CubeG0 (V1.4.0) will integrate the ``Version 10.3.1`` of ``CMSIS-FreeRTOS``. 

Thank you again for your interest. 

With regards,

ALABSTM, 2020-12-02:
Hi @bridadan,

STM32CubeG0 **v1.4.0** supporting FreeRTOS **v10.3.1** has just been published on GitHub. We hope this helps you solve the issue you reported here.

With regards,

closed 2020-12-02 by ALABSTM

bridadan, 2020-12-02:
Thanks for the update!