STMicroelectronics/STM32CubeG0 #11: stm32g0xx_ll_rtc.h compilation warning with GCC9

jeromecoutant, 2020-08-24:
Hi

There is a compilation warning with GCC v9.2.1 at this line:
https://github.com/STMicroelectronics/STM32CubeG0/blob/5114d9a57d8a450082fc8cafe4fe34d3610d0cef/Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_rtc.h#L3077

````
[Warning] stm32g0xx_ll_rtc.h@3077,3: 'register' is not at beginning of declaration [-Wold-style-declaration]
````

Correction is to remove "const"

Regards,

Jerome

RKOUSTM, 2020-08-24:
Hi @jeromecoutant,

Thank you for your contribution. The ``register`` keyword in the ``stm32g0xx_ll_rtc.h`` is removed. 
This update will be integrated in the future  firmware package release (``STM32G0 version v1.4.0``).

Thank you again for your contribution.

With regards,

RKOUSTM, 2020-08-24:
ST Internal Reference: 84398

closed 2020-12-02 by ALABSTM