STMicroelectronics/STM32CubeG0 #12: Undefined RCC_CFGR_SWS_* symbols in migrated project

bridadan, 2020-12-02:
**Describe the set-up**
 * The board (either ST RPN reference or your custom board): Custom board using a STM32G070KB
 * IDE or at least the compiler and its version: STM32Cube IDE v1.3.1

**Describe the bug**
After migrating a project from STM32CubeG0 1.3.1 (STM32Cube IDE was upgraded from 1.2.0) to 1.4.0, I attempted to build the project and received the same error as described here: https://community.st.com/s/question/0D53W00000OUZHcSAP/stm32cubefwg0v140-bug-

**How To Reproduce**
1. Create a project using STM32CubeG0 1.3.1
2. Migrate the project to STM32CubeG0 1.4.0
3. Run a "Clean"
4. Build (I built with the Debug profile)

ASELSTM, 2020-12-08:
Hi @bridadan, 

Thank you for your contribution. 

For a better understanding of the source of the issue may i ask you if you have used STM32CubeMx to generate the project or not ?

Best regards,

bridadan, 2020-12-08:
Hi @ASELSTM,

I have used the STM32Cube IDE project generator (File -> New -> STM32 Project), so I don't believe so. After upgrading the IDE and selecting the project's .ioc file, I was prompted to continue using the STM32Cube version the project was using (1.3.1), or to migrate to the latest version (1.4.0).

After checking the forum page I linked above, it looks like it may be as simple as deleting and regenerating `system_stm32g0xx.c`. I'll give that a try and report back here when I have time.

Thanks!

ASELSTM, 2020-12-08:
Hi @bridadan, 

Actually, this kind of project generation is associated to the STM32CubeMx. STM32Cube IDE project generator is based on STM32CubeMx tool which is in charge of the project generation.  

This issue has been already tracked internally and STM32CubeMx team is working on it. It will be fixed and made available in a future STM32CubeMX release.

In general terms, we don't treat aspects related to CubeMX tool at GitHub level, they are rather treated at ST Community level. 
Now, as this issue is not directly related to software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to ecosystem (CubeMX tool), please allow me to close it.

Thank you for your comprehension and thank you again for your contribution. We are looking forward to reading from you again.

With regards,

closed 2020-12-08 by ASELSTM

bridadan, 2020-12-08:
Thanks for clarifying @ASELSTM!