STMicroelectronics/STM32CubeG0 #13: HAL: USB_DRD_FS: content of stm32g0xx_hal_hcd.h should be included only if USB available.

str4t0m, 2020-12-15:
**Describe the bug**
The content of stm32g0xx_hal_hcd.h should be only included for devices where the USB peripheral is available.

**How To Reproduce**
1. In my project I have HAL_HCD_MODULE_ENABLED in  stm32g0xx_hal_conf.h
2. Compile the project with a device that does not support USB(e.g. stm32g071xx).
3. stm32g0xx_hal_hcd.h depends on defines in stm32g0XXxx.h that are only available if the device has an USB peripheral( e.g.: USB_DRD_TypeDef, USB_DRD_CfgTypeDef ...), resulting in an compile error.

**Expectation**
File content is included only if the peripheral USB_DRD_FS is defined:

`
#ifdef __cplusplus
extern "C" {
#endif

/* Includes ------------------------------------------------------------------*/
#include "stm32g0xx_ll_usb.h"

#if defined (USB_DRD_FS)

... The rest of the code  ...

#endif /* defined (USB_DRD_FS) */

#ifdef __cplusplus
}
#endif
`

**Additional context**
I am currently trying to update the HAL for STM32G0 Series for Zephyr to Version 1.4.0.
In Zephyr all HAL_PERIPHERALXX_ENABLED are defined in  stm32g0xx_hal_conf.h, therefore all devices from the STM32G0 Series missing the USB peripheral can't compile when this include guard is missing.

ALABSTM, 2021-01-20:
Hi @str4t0m,

Thank you for your contribution. Indeed, USB DRD-related definitions and declarations should be surrounded by pre-processor directives `#if defined (USB_DRD_FS)` and `#endif`. The issue will be logged into our internal database. A fix will be made available in the frame of a future release.

With regards,

ALABSTM, 2021-01-20:
ST Internal Reference: 99814

closed 2021-04-23 by RKOUSTM