STMicroelectronics/STM32CubeG0 #14: Sets incorrect SysTick frequency

raizub, 2021-03-11:
**Set-up**
 * STM32CubeIDE 1.6.0 with STM32G0 series 1.4.1 package.
 * MCU STM32G071C8

**The bug**
* Sets incorrect SysTick frequency. 

**How To Reproduce**
1. Generate main code with HSI RC, with PLLCLK to 60 MHz SYSCLK. 
2. SysTick frequency will be calculated from default 16 MHz value.

**Additional context**
Attached image with Initialization routine from current source.
I think, first should be called `LL_SetSystemCoreClock(60000000);` function to update `SystemCoreClock` variable, and only after this `HAL_InitTick (TICK_INT_PRIORITY)`, because this function uses `SystemCoreClock` to set correct frequency.

**Screenshots**
[image]

ALABSTM, 2021-03-23:
Hi @raizub,

Thank you for this report. Have you started from one of our provided examples or is it an application generated from scratch using Cube MX?

Thank you,

ALABSTM, 2021-03-23:
Hi @raizub,

I found the point you have reported is related to USB-PD applications, mainly [USB-PD_Consumer_1port](https://github.com/STMicroelectronics/STM32CubeG0/tree/master/Projects/STM32G0C1E-EV/Applications/USB-PD/USB-PD_Consumer_1port) and [USB-PD_Provider_1port](https://github.com/STMicroelectronics/STM32CubeG0/tree/master/Projects/STM32G0C1E-EV/Applications/USB-PD/USB-PD_Provider_1port) on the **STM32G0C1E-EV** board.

What you reported seems an issue indeed. It will be forwarded to our development teams for confirmation. I will get back to you as soon as I have their feedback. Thank you again for your contribution.

With regards,

rsystec, 2021-03-24:
> Hi @raizub,
> 
> Thank you for this report. Have you started from one of our provided examples or is it an application generated from scratch using Cube MX?
> 
> Thank you,

Hi,

The application was generated from scratch. But maybe not with the 1.4.1 package, but with some previous version (might be V1.4.0). Started using sysTick only after an update, so I noticed only then.

ALABSTM, 2021-05-05:
Hi @raizub,

I hope you are fine. Please excuse this somewhat delayed reply. Actually, this issue is related to the **Cube MX** tool generating the `SystemClock_Config()` function. This is a known issue that has been internally fixed and will be made available in a future release of the Cube MX. In the meanwhile, you shall manually invert the call order of the functions as you suggested. I hope this helps.

As this revealed to be an issue related to our ecosystem, not to the software delivered within this repository, please allow me to close it. Thank you for your comprehension and thank you again for having reported.

With regards,

ALABSTM, 2021-05-05:
ST Internal Reference: 100259

closed 2021-05-05 by ALABSTM