STMicroelectronics/STM32CubeG0 #17: HAL_SPI_TransmitReceive_IT argument misalignment

rardiol, 2021-08-16:
**Describe the set-up**
 * custom board with STM32G070KB
 * Compiler: ARM's gcc-arm-none-eabi-10-2020-q4-major
 * STM32CubeIDE
Version: 1.7.0
Build: 10852_20210715_0634 (UTC)
OS: Linux, v.5.8.0-63-generic, x86_64 / gtk 3.24.23
Java version: 11.0.10

**Describe the bug**

Hard Fault

**How To Reproduce**

Configure SPI for 8 bits mode using the .ioc file.

Declare two globals
```
uint8_t _spi_write[3];
uint8_t _spi_reading[3];
```

Call `HAL_SPI_TransmitReceive_IT(&hspi1, _spi_write, _spi_reading, 3) != HAL_OK`

Worked fine with -O0, but when switching to -Os I get a HardFault inside SPI_2linesRxISR_8BIT

**Additional context**

Further testing with `fsanitize=undefined` a crash happens inside the HAL code on what I think is an alignment check for `((uint16_t *)hspi->pTxBuffPtr)`. That pointer comes from `_spi_write`, which is a uint8_t array, so not 2 byte aligned. pTxBuffPtr is also a (uint8_t*) 

This seems to fix the problem:
```
uint8_t _spi_write[SPI_LENGTH] __attribute__ ((aligned(2)));
uint8_t _spi_reading[SPI_LENGTH] __attribute__ ((aligned(2)));
```

Is it documented anywhere that the arguments to the SPI HAL should be 2 byte aligned?

ASELSTM, 2021-09-02:
Hi @rardiol,

Thank you for your contribution. Unfortunately we were unable to reproduce the described behavior, in order to allow a better analysis of the problem, would you please share the whole project you have used to reproduce this issue.

With regards,

ASELSTM, 2021-10-07:
Hi @rardiol,

Please allow me to close this thread as no activity has been registered. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-10-07 by ASELSTM