STMicroelectronics/STM32CubeG0 #18: Issue with MicroLib

hattmt, 2021-09-20:
**Describe the set-up**
STM32G0 , Keil

**Describe the bug**
CubeMX set MicroLib unchecked at each generation.

**How To Reproduce**
Activate Microlib in Keil (project settings) and regenerate with CubeMX, .
Now go to the project settingds in Keil, it will be unchecked .

[image]
[image]

RKOUSTM, 2021-09-21:
Hi  @hattmt,

Thank you for your contribution.

In order to allow a better analysis of this problem, could you please give us more details about the application that you are using to reproduce this issue or did you use one of our applications? 

Also, could you please confirm you are using the latest firmware version and that the issue is systematically reproducible on your side?

Thank you again for your contribution.

With regards,

hattmt, 2021-09-21:
I use Keil(uVision5) for my project developpment  and STM32CubeMX to generate the HAL initialisation. I am programming on a Stm32G0.
I need MicroLib for my project. So I check the option in Keil to include MicroLib.

Until now, when I generated my project, CubeMX was keeping the Microlib checked in Keil settings at each generation , so it was working : CubeMX wasn't modifiying this setting.

In the new version of STM32CubeMX, each time I generate with CubeMx , it sets my Microlib option  uncheck in Keil. So here CubeMX goes and modify the value in my keil project and set it unchecked , I just need that Stm32CubeMx doesn't modify the settings that are set for microlib like the previous versions.

Regards.

RKOUSTM, 2021-09-22:
Hi @hattmt,

First we would like to thank you for your contribution.

The issue you pointed out is confirmed. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-09-22:
ST Internal Reference:  114126

closed 2021-09-22 by RKOUSTM