STMicroelectronics/STM32CubeG0 #19: I2C Slave examples for STM32G0316-DISCO board

kholia, 2021-09-25:
Problem: The STM32G031J6 MCU (on a STM32G0316-DISCO board) is unable to work properly as an I2C Slave when operating in the Interrupt Mode with callbacks functions.

Details:

[image]

This screenshot shows that the Slave is unable to handle `Data read` requests from the Master I2C device.

The Slave returned data is always `0xFF`. I saw https://www.i2c-bus.org/i2c-primer/analysing-obscure-problems/data-bytes-from-slave-are-0xff/ on this topic.

The I2 Master device is Arduino Pro Mini ATMega328P running @ 8 MHz @ 3.3v.

Note: I am using 2.2k pull-up resistorts on the SDA and SCL lines. Removing them doesn't make any difference.

Here is the configured pinout for my STM32G031J6 MCU:

[image]

I2C2 global interrupt is enabled in my STM32CubeIDE project.

Here is how the I2C pins are configured:

[image]

Clocks are configured as below:

[image]

Note: In comparison, when I run a similar STM32CubeIDE project on my STM32F103C8T6  "Blue Pill" board, things just work well as expected (Slave is able to send back actual data).

I am attaching the source code for the I2C Master and I2C Slave (STM32G031J6) below:

- [i2c_slave_code.zip](https://github.com/STMicroelectronics/STM32CubeG0/files/7229160/i2c_slave_code.zip)
- [master.ino.txt](https://github.com/STMicroelectronics/STM32CubeG0/files/7229159/master.ino.txt)

Questions:

- Is I2C Slave Mode (In `Interrupt Mode`) supported on STM32G031J6 MCU?

- Is there a reason why the STM32G0316-DISCO board software package has no `I2C` examples? They will be super helpful for users.

What further information may I provide for debugging this problem? Thanks for your help! :)

kholia, 2021-09-25:
Possibly related https://community.st.com thread: [Not able to read data from STM32G031 MCU I2C slave](https://community.st.com/s/question/0D53W00000Pnh5A/not-able-to-read-data-from-stm32g031-mcu-i2c-slave).

kholia, 2021-09-27:
Possibly related issue:

If I enable clock stretching on this STM32G0 MCU, the SCL line goes down, and remains low forever. This could be a bug in my code but very similar code runs perfectly fine on a "Blue Pill" STM32 board with clock stretching enabled.

KORKAD, 2021-12-10:
Hello @kholia,
Please note that similaire HAL I2C examples are already available for STM32G031K8 MCU but on Nucleo board not the DISCO.

For more details please refer to the I2C_TwoBoards_RestartComIT example. 

With Regards
KORKAD

kholia, 2021-12-13:
Thanks! This ticket may be closed.

closed 2021-12-14 by RKOUSTM