STMicroelectronics/STM32CubeG0 #24: cast conversion issue when _LL_RTC_CONVERT_BCD2BIN macro is used

jealcuna, 2021-11-24:
I am using stm32cube ide 1.7.0 and when I use the macro _LL_RTC_CONVERT_BCD2BIN a warning rise. I have wconversion warning flag enable, also, when I call the macro y use a uint8_t type data.

[image]

In the picture attached __VALUE__ is replaced by a uint8_t variable with a valir bcd number.

jealcuna, 2021-11-25:
A workaround for this is replace the line with:
#define __LL_RTC_CONVERT_BCD2BIN(__VALUE__) (uint8_t)((((uint8_t)((__VALUE__) & (uint8_t)0xF0) >> (uint8_t)0x04) * (uint8_t)10) + ((__VALUE__) & (uint8_t)0x0F))

ALABSTM, 2022-08-05:
Hi @jealcuna,

I hope you are fine. We tried to reproduce the problem you reported, with **no success**. Below the environment configuration used:
* STM32CubeG0 **v1.6.1**
* Cube IDE **v1.9.0**
* Optimization level `-O2`
* Compilation flags `-Wall` and `-Wconversion` enabled

Would you mind precising the types of variables `time_bcd` and `ptm.tm_hour` used in your code?

Thanks,

ALABSTM, 2023-04-10:
Hi @jealcuna,

I hope you are doing well. As there has been no activity on this thread since a while and as we could not reproduce the issue you reported, please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2023-04-10 by ALABSTM