STMicroelectronics/STM32CubeG0 #25: STM32CubeMX does not generate needed static library imports in Makefile

DinkDonk, 2021-11-25:
**Set-up**
 * NUCLEO-G0B1RE
 * STM32CubeMX with `Makefile` selected as `Toolchain / IDE`

**Description**
When setting up a USB-PD project – STM32CubeMX does not generate the needed static library imports in the Makefile, causing a lot of `undefined reference` errors when compiling. These references are impossible to look up due to them being defined in archived libraries, making it impossible to debug.

**How To Reproduce**
1. Set up a new project in STMCubeMX according to [AN5418](https://www.st.com/resource/en/application_note/dm00663511-how-to-build-a-simple-usbpd-sink-application-with-stm32cubemx-stmicroelectronics.pdf).

2. Choose `Makefile` as `Toolchain / IDE` in the `Project Manager` tab.

3. Generate the code and build with `$ make`

4. Observe failed build caused by `undefined reference` errors in the output.

**Solution**
The error is caused by missing static library definitions in the generated Makefile.

*Generated Makefile:*

```Makefile
# libraries
LIBS = -lc -lm -lnosys 
LIBDIR = 
LDFLAGS = $(MCU) -specs=nano.specs -T$(LDSCRIPT) $(LIBDIR) $(LIBS) -Wl,-Map=$(BUILD_DIR)/$(TARGET).map,--cref -Wl,--gc-sections
```

*Expected Makefile:*

```Makefile
# libraries
LIBS = -lc -lm -lnosys 
LIBDIR = Middlewares/ST/STM32_USBPD_Library/Core/lib/USBPDCORE_PD3_FULL_CM0PLUS_wc32.a
LDFLAGS = $(MCU) -specs=nano.specs -T$(LDSCRIPT) $(LIBDIR) $(LIBS) -Wl,-Map=$(BUILD_DIR)/$(TARGET).map,--cref -Wl,--gc-sections
```

Observe change in `LIBDIR`.  
The static library is chosen by selecting a `Stack Configuration` for the `USBPD` middleware in STM32CubeMX.  
Correct behaviour can be seen when doing the exact same project setup inside STM32CubeIDE.

RKOUSTM, 2021-12-08:
Hi @DinkDonk,

Thank you for this report. Unfortunately we don't treat aspect related to CubeMx tool in our GitHub repositories. Please allow me to redirect you to the [ST Community](https://community.st.com/s/question/0D53W00000JhsU4SAJ/stm32cubemx-ide-issue-throws-warning-when-using-macros). You can report this issue in the CubeMX related section. There you shall find people who will address your request and whom you will hopefully find a solution with.

Thank you again for your report. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2022-01-07:
Hi @DinkDonk,

An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release of CubeMX..

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2022-01-07:
Hi @DinkDonk,

An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release of CubeMX.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2022-01-07:
ST Internal Reference: 120440