STMicroelectronics/STM32CubeG0 #27: USB - PMA Buffer 0 ignored in isochronous transfers

marsam94, 2022-01-12:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * NUCLEO-G0B1RE
 * CubeIDE 1.8.0

**Describe the bug**
When using USBD Audio class, only one PMA buffer is being used (buffer 1) for OUT transfers. Resulting in only half packets being received.

**How To Reproduce**

- Enable USB in Device mode with pins PA11 and PA12

[image]

- Enable Audio Device Class with Frequency of 48KHz

- Generate code with CubeMX

- use malloc with USBD_malloc (usbd_conf.h)
- upsize heap size

As per default in USBD_Audio_DataOut, USBD_LL_GetRxDataSize is called to check the packet size received but it returns 0 then the packet size then back to 0. I added a buffer "packet_buf" for debugging purposes which gets the value returned on every call :
[image]

The PMA is configured this way : 
```
  HAL_PCDEx_PMAConfig((PCD_HandleTypeDef*)pdev->pData, 0x00, PCD_SNG_BUF, 0x18);
  HAL_PCDEx_PMAConfig((PCD_HandleTypeDef*)pdev->pData, 0x80, PCD_SNG_BUF, 0x58);
  HAL_PCDEx_PMAConfig((PCD_HandleTypeDef*)pdev->pData, AUDIO_OUT_EP, PCD_DBL_BUF, 0x01580098);
```

When looking directly inside the memory, we can see that only the values situated at the PMABuffer1 are changed :
[image]
[image]

If changing the addresses in PMAConfig, we can also see that it's always the address at pmaaddr0 that's ignored.

Thanks

ALABSTM, 2022-01-30:
Hi @marsam94,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

marsam94, 2022-07-27:
Hello, 
the new resolved my issues. Thanks for looking into it.

Sam

closed 2022-07-27 by marsam94