STMicroelectronics/STM32CubeG0 #31: I2C addr flag not reset inside I2C_ITAddrCplt

kjsdf7wr3, 2022-02-11:
The address flag is reset for
- 10bit mode (if event count == 2)
- listen is disabled

it is **not reset** for 
- 7bit mode
- 10bit mode (if event count == 1)

See
https://github.com/STMicroelectronics/STM32CubeG0/blob/03cb8e9ec0cbefad623caebe47359df0bab1d05e/Drivers/STM32G0xx_HAL_Driver/Src/stm32g0xx_hal_i2c.c#L5486

I think this is an error, it should be reset in all cases before calling `HAL_I2C_AddrCallback(hi2c, transferdirection, slaveaddrcode);`.
If the flag is not reset, the chip will pull the clock line low forever.

etheory, 2022-06-12:
Is this being actively looked into? Why was this fix not released in v1.5.1?
My G0 isn't working in slave mode because of what appears to be this bug.
It just hangs forever after sending the first data transfer successfully.

ASELSTM, 2023-04-10:
ST Internal Reference: 149407