STMicroelectronics/STM32CubeG0 #36: Please update FreeRTOS to the latest version

bojczmac, 2022-07-27:
When I enable link time optimization (-flto) I get a linker error regarding vTaskSwitchContext():

arm-none-eabi/bin/ld: /tmp/ccP992vo.ltrans0.ltrans.o: in function `PendSV_Handler':
<artificial>:(.text.PendSV_Handler+0x1c): undefined reference to `vTaskSwitchContext'

This issue was fixed in the latest version of the FreeRTOS so please update.

HBOSTM, 2022-12-21:
Hello @Bojczmac

Would you please give us more details about how you got this issue? Did you observed it at runtime ? if so, could you please share the code you have used to reproduce the problem.

With regards,

HBOSTM, 2023-04-13:
Hi @bojczmac ,

A newer version of this library is available with several bugfixes [here](https://github.com/FreeRTOS/FreeRTOS). Unfortunately, there is no plan in the near future to integrate any newer version of the library into this firmware. We will keep you informed, should there be any update in this plan.
In the meanwhile, you can easily integrate that newer version into your project by yourself, as it is backward compatible.
Please allow me to close this issue. Thank you for your comprehension and thank you for your report.

With regards,

closed 2023-04-13 by HBOSTM