STMicroelectronics/STM32CubeG0 #37: LL Syscfg comments incorrectly identify COMP EXTI lines

grantg012, 2022-08-17:
The comments in https://github.com/STMicroelectronics/STM32CubeG0/blob/master/Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_system.h#L1021
incorrectly label COMP1 and COMP2 to use EXTI 21 and 22 respectively. As per the reference manual, they use EXTI 17 and 18.

**Additional context**
If you have a first analysis or a patch proposal, thank you to share your proposal.

**Screenshots**
[image]

ASELSTM, 2022-09-01:
Hi @grantg012,

The comment are indeed incorrect.  The EXTI lines should be updated as you have mentioned. 

```diff
/**
-  * @brief  Check if Comparator 1 interrupt occurred or not (EXTI line 21).
+  * @brief  Check if Comparator 1 interrupt occurred or not (EXTI line 17).
  * @rmtoll SYSCFG_ITLINE12 SR_COMP1      LL_SYSCFG_IsActiveFlag_COMP1
  * @retval State of bit (1 or 0).
  */
```

```diff
/**
-  * @brief  Check if Comparator 2 interrupt occurred or not (EXTI line 22).
+  * @brief  Check if Comparator 2 interrupt occurred or not (EXTI line 18).
  * @rmtoll SYSCFG_ITLINE12 SR_COMP2      LL_SYSCFG_IsActiveFlag_COMP2
  * @retval State of bit (1 or 0).
  */
  */
```

With regards,

TOUNSTM, 2023-11-23:
fixed in commit 9f4c4a1c1ee2a497eb94defee1525f8cf7100bd5

closed 2023-11-23 by TOUNSTM