STMicroelectronics/STM32CubeG0 #40: rand() function uses 5K space

eos1d3, 2023-02-15:
**Describe the set-up**
 * STM32G030C8Tx
 * STM32CUBEIDE

**Describe the bug**
A clear and concise description of what the bug is.

**How To Reproduce**
1. add `#include <stdlib.h>`

2. Call` srand() `and also `rand()`

3. compile the project to see the size

4. remove `srand()` and `rand()` and compile to compare the size

**Additional context**
I found `vprintf`, `printf`, `putc`, `puts`... are linked. But none of them are used in the project. Once the random functions are removed, all of these are gone.

HBOSTM, 2023-03-01:
Hello @eos1d3 ,

From my side there is no problem in the project size with random functions, Would you please give us more details about how you got this issue? And share the code you have used to reproduce this issue.

Best Regards,

eos1d3, 2023-03-01:
Hi, 

Just use STM32CubeIDE to create a empty project. I use STM32G030C8T.

Add the header:

`#include "stdlib.h"`

And the `rand()` function in main loop:

`rand();`

Comment out the 'rand()' call to compare the code size.

With `Release` config:

No `rand()` call:
```
   text	   data	    bss	    dec	    hex	filename
   2912	     20	   1572	   4504	   1198	stm32_exe.elf
```

Has `rand()` call:
```
arm-none-eabi-objdump -h -S  stm32_exe.elf  > "stm32_exe.list"
   text	   data	    bss	    dec	    hex	filename
   7904	    120	   1592	   9616	   2590	stm32_exe.elf
```

My IDE version:
STM32 Cube FW_G0 V1/6/1
STM32CubeIDE Version: 1.11.2 for Mac (previous version also has the same issue)
Project settings: everything is default

eos1d3, 2023-05-18:
No fix?

Actually I use the following to fix the problem

```
void Random::srand(unsigned int seed) {
	next = seed;
}

int Random::rand(void) // RAND_MAX assumed to be 32767
		{
	next = next * 1103515245 + 12345;
	return (unsigned int) (next / 65536) % 32768;
}

```

HBOSTM, 2023-05-29:
Hi @eos1d3,

Thank you again for this contribution, this point has been reported to our development teams. I will get back to you as soon as I have any updates.

Best Regards,

HBOSTM, 2023-05-29:
ST Internal Reference: 153499

HBOSTM, 2023-07-17:
Hi @eos1d3 ,

the extra amount of code is due to newlib-nano reentrancy checking, you can either use standard newlib in that particular case rand() will be smaller and not import printf, etc... but the overall size of the lib contribution will be bigger, or you can keep using nano and define the assert function like this:

void __assert_func(const char *, int, const char *, const char *) {
    while(1);
}

which will avoid pulling printf symbols, and don't forget to add -Wl,--gc-sections flags to remove unused symbols (if you didn't).

Best Regards,
Houssine

HBOSTM, 2023-07-28:
Hello @eos1d3,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve this issue. Thank you for your comprehension.

With regards,

closed 2023-07-28 by HBOSTM