STMicroelectronics/STM32CubeG0 #42: LL Function for checking UCPD RxErr Flag is missing

ualbertagreen, 2023-03-01:
The function for checking is the RxErr flag is missing. Despite the fact the RxErr flag is defined in LL-UCPD, it is never used.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

The function for checking RxErr flag should look like this:
```
/**
  * @brief  Check if Rx error interrupt
  * @rmtoll SR          RXERR         LL_UCPD_IsActiveFlag_RxErr
  * @param  UCPDx UCPD Instance
  * @retval None
  */
__STATIC_INLINE uint32_t LL_UCPD_IsActiveFlag_RxErr(UCPD_TypeDef const * const UCPDx)
{
  return ((READ_BIT(UCPDx->SR, UCPD_SR_RXERR) == UCPD_SR_RXERR) ? 1UL : 0UL);
}
```

ALABSTM, 2024-02-08:
Hi @ualbertagreen,

Please excuse this delayed reply. Your request looks very legitimate. It has been forwarded to our development teams. I will keep you posted.

With regards,

ALABSTM, 2024-02-08:
ST Internal Reference: 173010