STMicroelectronics/STM32CubeG0 #43: missing linker description file *.ld

MrMfromK, 2023-03-09:
Hello,
great to find everything here on github in a well ordered structure.
When creating a project with CubeMx I always get a *.ld for the chosen MCU copied into the project. But I could nor find it on github.
Is it possible to include this? (Or did I miss it?)
Thanks
Martin

HBOSTM, 2023-03-15:
Hello @MrMfromK ,

First of all, thank you for this contribution.
After checking, all the linker files already exist in the GitHub STM32CubeIDE projects, please reload again the Cube Firmware.
So please allow me to close this issue.

Best Regards,

MrMfromK, 2023-03-15:
Hello HBOSTM,
thanks for your answer.
I am sorry but still after looking through the whol directory tree on https://github.com/STMicroelectronics/STM32CubeG0 under code, I still can't find a linker description file. I know that I can create one by creating a project with CubeMx, but what I want is to download it directly from the named github link. Please point me to the place where to find it.
Thanks again
Martin

HBOSTM, 2023-03-15:
Hello @MrMfromK ,

Thank you again for this contribution .
The linker files exist under STM32CubeIDE projects, under all Examples like 
[CRC_Bytes_Stream_7bit_CRC](https://github.com/STMicroelectronics/STM32CubeG0/tree/master/Projects/NUCLEO-G031K8/Examples/CRC/CRC_Bytes_Stream_7bit_CRC/STM32CubeIDE) example 

Best Regards,

MrMfromK, 2023-03-15:
Ok, thanks for the quick answer. Now I found it.
Things are now clearer to me. It seems the .ld-file is generated by CubeMx when generating code for a project. That is why they are included in the examples. To get started I created a CubeIDE for my target controller which is a STM32G0B0VETX, generated code for it and took the .ld file from there. It is named STM32G0B0VETX_FLASH.ld. As there are no examples for this controller under STM32CubeG0/Projects I think the file can't be found there (still I didn't browse all examples ;-) )
So again I want to suggest to place a basic linker description file for each controller in a special folder maybe under the drivers folder. 
What do you think about that?
best regards
Martin

HBOSTM, 2023-04-12:
Hello @MrMfromK ,

Thank you for this contribution. This point has been reported to our development teams. I will get back to you as soon as I have any updates.

With regards,

TOUNSTM, 2023-11-23:
Hello @MrMfromK,

First, allow me to thank you for your contribution. Indeed, the GCC linker files has only been made for recent products such as H7, WB/WL, L5... because these series have brought some specificities (Dual core, secure / non secure portioning). However, no plan for our technical team to add GCC linker files for legacy products like G0.

Please, allow me to close this issue. Thank you once again for your contribution and thank you for your comprehension.

With regards,

closed 2023-11-23 by TOUNSTM