STMicroelectronics/STM32CubeG0 #44: Delay missing from ADC calibration

jj-uk, 2023-04-23:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
Custom board with a STM32G031G8.
IDE: STM32CubeIDE V1.11.2
Firmware Package: STM32Cube FW_G0 V1.6.1

**Describe the bug**
A clear and concise description of what the bug is.
HAL_ADCEx_Calibration_Start(), line 169 needs a delay to allow the ADC to be fully enabled BEFORE calling LL_ADC_SetCalibrationFactor().
This can be achieved by polling/waiting for LL_ADC_IsActiveFlag_ADRDY(hadc->Instance) to return true, although this doesn't apply a timeout, so could end in deadlock. A polled solution with a timeout should be implemented, as seen as used elsewhere in this function.

**How To Reproduce**
1. Indicate the global behavior of your application project.
Simple project reading 4 ADC channels, storing results via DMA channel1.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
stm32g0xx_hal_adc_ex.c
https://github.com/STMicroelectronics/STM32CubeG0/blob/master/Drivers/STM32G0xx_HAL_Driver/Src/stm32g0xx_hal_adc_ex.c#L169

3. The use case that generates the problem.
Calling HAL_ADCEx_Calibration_Start()

4. How we can reproduce the problem.
Create an application that uses the ADC (HAL).
Call HAL_ADCEx_Calibration_Start() - This will return an error.

**Additional context**
This bug has been known about for quite some time, but not fixed.
https://community.st.com/s/question/0D53W00001aJM92SAG/stm32cubewlv120-haladcexcalibrationstart-does-not-work
See comment dated September 21, 2022 at 10:31 PM.

**Screenshots**
If applicable, add screenshots to help explain your problem.

TOUNSTM, 2023-04-25:
Hello @jj-uk,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

RJMSTM, 2023-07-27:
Hello @jj-uk,

Thank you for your report. An internal tracker has been logged and a fix was implemented and made available in the frame of a STM32G0 future release.

Thank you again for you report.

With regards,
Rania

jj-uk, 2024-02-26:
When will this be released?

RJMSTM, 2024-02-27:
Hello @jj-uk,

The issue has been fixed in the frame of version v1.6.2 of the STM32CubeG0 published on GitHub Dec 22, 2023.

 Thank you again for having reported.

closed 2024-02-27 by RJMSTM

insolace, 2025-06-23:
Sorry to reopen an old issue, but the fix to this problem assumes that the ADC is using the system clock. If RCC.ADCCLockSelection=RCC_ADCCLKSOURCE_HSI and the prescaller is DIV4, then the delay is not applied and the ADC fails to run.

See line 179 of stm32g0xx_hal_adc_ex.c:

if (adc_clk_async_presc >= LL_ADC_CLOCK_ASYNC_DIV16)