STMicroelectronics/STM32CubeG0 #46: The `HAL_I2C_IsDeviceReady()` may return an error, even if the I2C device is OK

chuyec, 2023-12-29:
**Describe the set-up**

 * STM32G070RBT based custom board.
 * STM32CubeIDE 1.14.0.
 * STM32Cube FW_G0 V1.6.2
 *  Use I2C1 or I2C2 to communicate with CAT24C02TDI EEPROM memory

**Describe the bug**

When checking an I2C device, the check function `HAL_I2C_IsDeviceReady()` may return an error, even if the I2C device is OK.

The _STM32Cube FW_G0 V1.6.1_ does not have such problem.

**How To Reproduce**

Call the `HAL_I2C_IsDeviceReady()` function several times:
```
HAL_I2C_IsDeviceReady(eeprom->hi2c, eeprom->dev_addr, 100, 200);
```

**Additional context**

Supposedly, the problem is in `I2C_IsErrorOccurred()` function.

If the next condition is true:
```
  if (HAL_IS_BIT_SET(itflag, I2C_FLAG_AF))
```
the next code will always be executed:
```
    error_code |= HAL_I2C_ERROR_AF;

    status = HAL_ERROR;
```
And then the `I2C_IsErrorOccurred()` function will fail.

ALABSTM, 2024-01-02:
Hi @chuyec,

First things first. Best wishes for this new year 2024.

Thank you for this report. When checking function `I2C_IsErrorOccurred()`, I see **no change** between firmware versions **1.6.1** and **1.6.2**. I notice you used the expression "**may** return an error". Did you actually experienced an error or is it a guess? Thank you in advance for the clarification.

With regards,

chuyec, 2024-01-06:
Hi @ALABSTM ,

Yes, I actually experienced an error on version **1.6.2**. The error occurs sometimes. The error does not occur on version **1.6.1**.

Yes, indeed, there are no changes in the `I2C_Is Error Occurred()` function. **This is just my guess**.

But the next code was removed from `HAL_I2C_IsDeviceReady()`: 
[image]

And some code was added to the `I2C_WaitOnFlagUntilTimeout()` function: https://github.com/STMicroelectronics/stm32g0xx_hal_driver/blame/e194e5abc3a125ef08434f1ee122a2bc7b3d9af4/Src/stm32g0xx_hal_i2c.c#L6932
[image]

The call stack:
- `HAL_I2C_IsDeviceReady()`
    - `I2C_WaitOnFlagUntilTimeout()`
        - `I2C_Is Error Occurred()`

chuyec, 2024-02-20:
Hi @ALABSTM 

We solved the problem

Our workflow:

1. The STM32 I2C peripheral and I2C chip are working fine
2. Turning off the power to the I2C chip (STM32 I2C peripheral stays active)
3. Turning on the power to the I2C chip
4. `HAL_I2C_IsDeviceReady()` may return fail
5. If `HAL_I2C_IsDeviceReady()` was failed at least once then only `I2C_Deinit()` folowed by `I2C_Init()` can solve the problem

Now we executes `I2C_Deinit()`/`I2C_Init()` on every I2C chip power switching.

Yes, I agree that power switching of I2C chip that result to ground the I2C pins is not normal operation. But the **1.6.1** version handles this case better than **1.6.2**.
Please close the bug if you think this is ok.

Thank you for your attention

jmag9999, 2024-03-26:
I am seeing this issue also. The above workaround does solve the problem, but this should really be fixed properly.

KRASTM, 2024-06-04:
Hello @chuyec,

Please allow me to close this thread as no activity. You may reopen it at any time if you have any details to share with us. Thank you for your comprehension.

Best Regards.

closed 2024-06-04 by KRASTM