STMicroelectronics/STM32CubeG0 #47: stm32g0xx_hal_fdcan.c undocumented change to 'pTxHeader->DataLength'

ryedwards, 2024-01-17:
When updating a project to the latest HAL I was getting a hard fault when trying to TX a CAN message with my exisiting code base.

The differences exisit between 1.6.1 and 1.6.2.  A diff between the files confirms.

Reviewing the call stack the program was failing at this command:
/* Write Tx payload to the message RAM */
  for (ByteCounter = 0; ByteCounter < DLCtoBytes[pTxHeader->DataLength]; ByteCounter += 4U)

This was failing because the DataLength was out of range.

I went back to a saved version of my code to compare that area of code.  In the previous version this is what appears:
/* Write Tx payload to the message RAM */
  for (ByteCounter = 0; ByteCounter < DLCtoBytes[pTxHeader->DataLength >> 16U]; ByteCounter += 4U)

It appears between the two versions the DLC was changed from the shifted value to the absolute value.  This should be documented as it will eventually impact all of the DLC areas of my code.

RJMSTM, 2024-01-23:
Hello @ryedwards,

Firstly, I confirm that there is an error in the release note, as the message for this change is missing. We apologize for any inconvenience this may have caused, this will be fixed and integrated ASAP.

Second, in regard to the hard fault that was detected when you were trying to TX a CAN message, have you updated the header file because the change is also in stm32g0_hal_fdan.h
[image]

Regards,
Rania

RJMSTM, 2024-01-23:
Fixed in 11bf550bba048e55fb465842fa0b2d1f13660868

closed 2024-01-23 by RJMSTM