STMicroelectronics/STM32CubeG0 #49: Feature proposition: half-duplex / one-wire mode support

loulecrivain, 2024-06-26:
# Feature proposition: 1-wire protocol support
**Setup**
Should be any board supported, as described in the [README.md](https://github.com/STMicroelectronics/STM32CubeG0/blob/master/README.md) of this repository.

**Additional context**
Goal: support standard 1-wire for bootloader communication (with correct USART config - ideally no need for external active components).

**Comments ?**
I'd be glad to receive comments about this feature proposition. Would it be something welcome regarding the scope of this project ?

KRASTM, 2024-06-28:
Hello @loulecrivain,

Thank you for the report.

In fact, the driver supports this feature with UART mode, there is an API `HAL_StatusTypeDef HAL_HalfDuplex_Init()` to enable Half-Duplex mode.

With regards,

KRASTM, 2024-07-05:
Hello @loulecrivain,

Please allow me to close this issue, thank you for your comprehension.

With regards,

closed 2024-07-05 by KRASTM