STMicroelectronics/STM32CubeG0 #51: not usefull const in stm32g0xx_ll_dma.h

RetoFx, 2024-07-11:
const is one of the importers keywords in C/C++.
It is great, that you add more and more const.

Const tell the compiler, that the symbol is not writable.

For example,
Get function should set const for the object
Set function should not set const for the object

But now you add const to object that are modified.

For example,
__STATIC_INLINE void LL_DMA_EnableChannel(const DMA_TypeDef *DMAx, uint32_t Channel)
{
  uint32_t dma_base_addr = (uint32_t)DMAx;
  SET_BIT(((DMA_Channel_TypeDef *)(dma_base_addr + CHANNEL_OFFSET_TAB[Channel]))->CCR, DMA_CCR_EN);
}

This function modifies *DMAx. So DMAx should not be const.

Here it works because you cast DMAx to a uint32_t and so you lost the const attribute.
Casting is always a dangerous modification!

Please remove const from DMA_TypeDef *DMAx in the following functions
LL_DMA_EnableChannel()
LL_DMA_DisableChannel()
LL_DMA_ConfigTransfer()
LL_DMA_SetDataTransferDirection()
LL_DMA_SetMode()
LL_DMA_SetPeriphIncMode()
LL_DMA_SetMemoryIncMode()
LL_DMA_SetMemoryIncMode()
LL_DMA_SetMemorySize()
LL_DMA_SetChannelPriorityLevel()
LL_DMA_SetDataLength()
LL_DMA_ConfigAddresses()
LL_DMA_SetMemoryAddress()
LL_DMA_SetPeriphAddress()
LL_DMA_SetM2MSrcAddress()
LL_DMA_SetM2MDstAddress()
LL_DMA_SetPeriphRequest ()
LL_DMA_EnableIT_TC ()
LL_DMA_EnableIT_HT()

BTW. It is best practice when the pointer to an object is const.

For example,
DMA_TypeDef *DMAx
Should be replaced with 
DMA_TypeDef * const DMAx
So the object can modified but the pointer can’t be modified.

RetoFx, 2024-07-11:
Similare in stm32g0xx_ll_dmamux.h

RJMSTM, 2024-07-15:
ST Internal Reference: 186542

RJMSTM, 2024-07-17:
Hello @RetoFx,

it's ok for stm32g0xx_ll_dmamux.h for examples for the following functions:

LL_DMAMUX_SetRequestID()
LL_DMAMUX_SetSyncRequestNb()
LL_DMAMUX_SetSyncPolarity()
LL_DMAMUX_EnableEventGeneration()
LL_DMAMUX_DisableEventGeneration()
LL_DMAMUX_EnableSync()
LL_DMAMUX_DisableSync()
LL_DMAMUX_SetSyncID()
LL_DMAMUX_EnableIT_SO()
LL_DMAMUX_DisableIT_SO()

For stm32g0xx_ll_dma.h:

In all these functions, DMAx is not modified.

The conversion (uint32_t)DMAx is used to obtain a numerical representation of the base address of the DMA peripheral, which allows calculating the address of the specific DMA_Channel. This address is then used to access and modify the CCR, CPAR, CMAR, or CNDTR register of the DMA_Channel that already exists in the DMA_Channel_TypeDef structure.

```
typedef struct
{
  __IO uint32_t CCR;         /*!< DMA channel x configuration register        */
  __IO uint32_t CNDTR;       /*!< DMA channel x number of data register       */
  __IO uint32_t CPAR;        /*!< DMA channel x peripheral address register   */
  __IO uint32_t CMAR;        /*!< DMA channel x memory address register       */
} DMA_Channel_TypeDef;
```

Therefore, in this case, the pointer DMAx is read-only (const).

Regards,

closed 2024-08-07 by RJMSTM