STMicroelectronics/STM32CubeG0 #53: USBPD Analyzer often lags with I2C timeout

tom-van, 2025-01-26:
Two FreeRTOS threads call at least BSP_PWRMON_GetVoltage() without locking I2C by a mutex.
When another I2C operation is re-entered while the first one is not finished, I2C fails and waits till 3 sec timeout.
Mainly happens at https://github.com/STMicroelectronics/STM32CubeG0/blob/fcc63f394b69d25d005c431be60960d2c22acab8/Projects/STM32G071B-DISCO/Demonstrations/USBPD_Analyzer/Src/demo_disco.c#L1347

Mutex guarded version BSP_USBPD_PWR_VBUSGetVoltage() should be used instead of BSP_PWRMON_GetVoltage(ALERT_VBUS, ...).
Similarly BSP_USBPD_PWR_VBUSGetCurrent() should replace BSP_PWRMON_GetCurrent(ALERT_VBUS, ...).
Also other I2C should be guarded similarly, e.g. BSP_PWRMON_GetVoltage(ALERT_CC1 / ALERT_CC2 ...) but there is no direct replacement in BSP_USBPD_PWR_ module.

ALABSTM, 2025-01-30:
Hi @tom-van,

Your point looks relevant. Let's ask the development teams and see their feedback.

With regards,

ALABSTM, 2025-01-30:
ST Internal Reference: 201898

tom-van, 2025-01-30:
After writing this issue I found that in spy mode the mutex is probably not initialized and therefore BSP_USBPD_PWR_VBUSGetVoltage() doesn't work. I used condition similar to
https://github.com/STMicroelectronics/STM32CubeG0/blob/fcc63f394b69d25d005c431be60960d2c22acab8/Projects/STM32G071B-DISCO/Demonstrations/USBPD_Analyzer/Src/demo_disco.c#L1241
however I didn't go to the depth code to find if mutex locking is or isn't necessary in spy mode. Perhaps collisions are not so frequent in spy mode. I would recommend use a mutex always.